heishamon/HeishaMon #646: Listen only (parallel CZ-TAW1) not working on Panasonic M model

TonCoronel, 2025-02-07:
Hi,

I have replaced my Panasonic J serie with a Panasonic M serie

I have done a factory reset on the HeishaMon small and made sure I am running firmware 3.9 as it should be M series compatible.

I am using the HeishaMon purely for monitoring, CZ-TAW1 is used primarily in combination with external party managing the heatpump.

With the J serie when I select the "Listen Only" function it would work perfectly and continue to provide MQTT information to my Home Assistant.
With the M serie when I select  the "Listen Only" function it stops working, completely it seems, at least the "|Correct received data" goes to 0, no values are updated on the webpage of the Heishamon and also on the console no information of the heatpump is displayed. 

When I de-select the "Listen Only" function all information is received again and pushed to Home Assistant, however I am not certain what the impact is of having both the CZ-TAW1 and HeishaMon connected and both thinking they are the primary controller?

IgorYbema, 2025-02-07:
Can you make a photo on how you did connect them to the heatpump?
Maybe the M-series has both connections seperated now so both can work idependent

TonCoronel, 2025-02-12:
Yes sure, here are the pictures of the connections.
The CZ-TAW1 is connected to the CN-CNT2 (RED) and the Heishamon is connected to the CN NMODE

[image]
[image]

IgorYbema, 2025-02-12:
I think indeed CN-NMODE shares with the white CN-CNT and the red CN-CNT(2) is a seperated connection. That is why listen only isn't needed in this setup and not working also as it doesn't see the request from the CZ-TAW1 which is on red.

closed 2025-03-20 by github-actions[bot]