heishamon/HeishaMon #647: Panasonic series G

elvisquaglia, 2025-02-08:
Hi, I have a Panasonic series G modello WH-MHF09G3E5. This  model is compatibile whit your pcb and software?

MiG-41, 2025-02-08:
Probably it is same situation like for "F" series...
https://github.com/Egyras/HeishaMon/issues/229
https://github.com/v-s-c-o-p-e/geisha_aquarea_panasonic_arduino_esp8266_proxy
And probably there is a CN-CMODE connector on PCB , but no further documentation.

elvisquaglia, 2025-02-08:
> Probabilmente è la stessa situazione della serie "F"... [#229 ](https://github.com/Egyras/HeishaMon/issues/229) [https://github.com/vscope/geisha_aquarea_panasonic_arduino_esp8266_proxy](https://github.com/v-s-c-o-p-e/geisha_aquarea_panasonic_arduino_esp8266_proxy) E probabilmente c'è un connettore CN-CMODE sul PCB, ma non c'è ulteriore documentazione.

thanks for the reply, I looked but the project has been stuck for 5 years. do you know other solutions? I contacted panasonic but they say there would be a way to connect to the network but it seems very complex, they will let me know in the next few weeks. but if I look on my pump what connector I have could it be of help?

MiG-41, 2025-02-08:
First case is the connection to theHP ( via serial interface , i.e. USB-TTL ). For this have to have a litle "engineering skills" , know what i GND, RX ,TX , soldering perhaps... ect.... 
Next case is what to do after connect. HP generally not send anything without sending to it proper command ( here it was called MP - Magic Packet). 
If you send wrong - No Answer from HP....
We found it sniffing communication beetween HP and internet module.  
So either you could try to send standard MP from H and above and have luck ( or MP from F work as well) , or you buy this internet module to sniff , or you find someone who have it... 
When you find it , and it wary from H/G/L/M , then no one will make you firmware for Heishamon to fit ( not for one person...) ,you will have to do it your own....

Buying a ready Heishamon is a big risk ,that it will not work anyway.

closed 2025-03-16 by github-actions[bot]