heishamon/HeishaMon #652: Round function not working correctly, it's rounding to the nearest integer in the direction of 0

blb4github, 2025-02-10:
If I'm not mistaken the Round function is not working as it should, it's routing to the nearest integer in the direction of 0

```
on System#Boot then
  #Test = -3;
  setTimer(3,10);
end

on timer=3 then
	setTimer(3,10);
	#Test = #Test + 0.2;
	$Test1 = #Test;
	$Test2 = round($Test1);
end
```

The log output gives:

```

0   $Test1 = -2.8
1   $Test2 = -2                                should have been -3
1   $Test1 = -2.59999
2   $Test2 = -2                                should have been -3
1   $Test1 = -2.39999
2   $Test2 = -2                                correct
1   $Test1 = -2.19998
2   $Test2 = -2                                correct
1   $Test1 = -1.99998
2   $Test2 = -1                                should have been -2
1   $Test1 = -1.79998
2   $Test2 = -1                                should have been -2
1   $Test1 = -1.59998
2   $Test2 = -1                                should have been -2
1   $Test1 = -1.39998
2   $Test2 = -1                                correct
1   $Test1 = -1.19998
2   $Test2 = -1                                correct
1   $Test1 = -0.999977
2   $Test2 = 0                                should have been -1
1   $Test1 = -0.799976
2   $Test2 = 0                                should have been -1
1   $Test1 = -0.599976
2   $Test2 = 0                                should have been -1
1   $Test1 = -0.399976
2   $Test2 = 0                                correct
1   $Test1 = -0.199976
2   $Test2 = 0                                correct
1   $Test1 = 2.38419e-05
2   $Test2 = 0                                correct
1   $Test1 = 0.200024
2   $Test2 = 0                                correct
1   $Test1 = 0.400023
2   $Test2 = 0                                correct
1   $Test1 = 0.600023
2   $Test2 = 0                                should have been 1
1   $Test1 = 0.800022
2   $Test2 = 0                                should have been 1
1   $Test1 = 1.00002
2   $Test2 = 1                                correct
1    $Test1 = 1.20002
2   $Test2 = 1                                correct
1   $Test1 = 1.40001
2   $Test2 = 1                                correct
1   $Test1 = 1.60001
2   $Test2 = 1                                should have been 2
1   $Test1 = 1.80001
2   $Test2 = 1                                should have been 2
1   $Test1 = 2
2   $Test2 = 2                                correct
1   $Test1 = 2.2
2   $Test2 = 2                                correct
1   $Test1 = 2.39999
2   $Test2 = 2                                correct
1   $Test1 = 2.59999
2   $Test2 = 2                                should have been 3
```

blb4github, 2025-02-11:
Currently using work-around floor(X + 0.5)

CurlyMoo, 2025-02-11:
Good catch.
https://github.com/IgorYbema/HeishaMon/pull/154

McMagellan, 2025-02-12:
See closed issue #496 
Formatting a float number in rules

closed 2025-03-20 by github-actions[bot]

blb4github, 2025-10-01:
I believe this error is still in the latest official firmware. It would be great if this can be fixed in a new release.