heishamon/HeishaMon #654: Error parsing rules with function parameter passing

McMagellan, 2025-02-12:
I have now used the function with parameter passing in Rules for the first time. When checking why the rule was not accepted, I found the following.
Here is a test example with the rule that is published in your documentation. I added two lines at the end, see picture.

[image]

This rule is not accepted with the error: ERROR: Unexpected token (3340).
1) If I remove the parameter passing the rule is accepted. (on calcWar then)
2) If I swap the last two commandlines the rule is accepted.

I can say that if the function is called with parameters, the last command in the funktion must not be a settimer- command.
I use Heishamon V5 with firmware 3.9

McMagellan, 2025-02-12:
Example 1: If the line 
**(on calcWar($Ta1, $Tb1, $Ta2, $Tb2) then)** 
is changed to 
**(on calcWar then)**
the rule is accepted. However, the parameter transfer cannot then work.

[image]

Example 2: If the command
**(settimer(11,20);)**
is **not the last line of the rule**, the rule is accepted.

[image]

You can find the original faulty code here.
```
on calcWar($Ta1, $Tb1, $Ta2, $Tb2) then
	#maxTa = $Ta1;

	if @Outside_Temp >= $Tb1 then
		#maxTa = $Ta1;
	elseif @Outside_Temp <= $Tb2 then
		#maxTa = $Ta2;
	else
		#maxTa = $Ta1 + (($Tb1 - @Outside_Temp) * ($Ta2 - $Ta1) / ($Tb1 - $Tb2));
	end

    #Bonus = $NeuBonus;
    settimer(11,20);

end
```

blb4github, 2025-03-06:
@McMagellan. great you found what is working and what not, I do also experience sometimes such errors and indeed, sometimes changing the position of a line or swapping a rule to another position (e.g. swap rule 3 & 4) does solve the issue.

closed 2025-04-12 by github-actions[bot]