heishamon/HeishaMon #655: Byte 102 and 103

ElVit, 2025-02-18:
In ProtocolByteDecrypt.md Byte 102 and 103 are marked as "0 byte".
But my Heatpump returns 0x7E for byte 102 and 0x7C for byte 103. 

Hy Heatpump: IDU:WH-SDC0709J3E5, ODU:WH-UD09JE5

Here the complete message:
71,C8,01,10,56,55,62,89,00,55,00,01,00,00,00,00,00,00,00,00,26,15,11,55,9A,95,55,55,65,19,00,00,00,00,00,00,00,00,85,7B,80,80,AF,71,71,97,99,00,00,00,00,00,00,00,00,00,00,00,80,80,15,8A,85,85,D0,7B,78,1F,7E,1F,1F,79,79,8D,8D,B0,A3,76,8F,B7,A3,7B,8F,91,85,76,8F,8A,94,9E,8F,8A,94,9E,85,8F,8A,0D,5B,78,C1,0B,7E,7C,1F,7C,7E,00,00,00,55,55,55,21,A9,15,69,05,14,12,65,00,00,00,00,00,00,00,00,62,D2,0B,43,54,42,D2,0B,72,66,AD,00,A7,83,AC,AC,AD,32,AF,AF,AF,32,32,85,B1,AC,A8,9E,A7,81,80,88,80,85,27,01,01,01,00,00,22,00,01,01,01,01,79,79,C9,01,22,13,00,8C,07,00,01,00,00,01,00,00,0A,01,01,01,01,01,01,01,02,00,00,1A

Maybe this will help with decoding.

MiG-41, 2025-02-19:
Maybe they are dublicating , byte 102 with 106 , and byte 103 with 105... Could you change the deltas for heater , and see ...

Generally someone with J series have to decode it,since in H wersion there are zeros ....

ElVit, 2025-02-24:
I have tried to change byte 105 and 106 but somehow there isn't changing anything for me.
I'm using the [ESPHome component](https://github.com/ElVit/esphome_components/tree/main/components/panasonic_heatpump) which I created.
Unfortunatelly I dont know the valid range for TOP97 and TOP98, so maybe I write wrong values?
After changing Byte 106 it changed first to 0x80 but now I can not change it back to 0x7E.

Can you maybe tell me where to change TOP97 and TOP98 in the Heatpump touch panel?

closed 2025-04-01 by github-actions[bot]