heishamon/HeishaMon #657: Settings page stuck in "Please wait, loading saved settings..."

bemmbix, 2025-03-09:
When I call up the settings page (http://heishamon.lan/settings)
the settings page only appears once when the device has been reset to the default state.

Afterwards I only get a "Please wait, loading saved settings..." message.

Firmware Version 3.9
Tested with Firefox, Chrome under Linux and Windows... same result.

McMagellan, 2025-03-10:
Have you tried using a "private tab"?

bemmbix, 2025-03-10:
Yes, as suggested in #473

bemmbix, 2025-03-10:
The getsettings page give me all the settings in a JSON-String. So, I think that it works  in the background.

When starting the F12 Debug tool it shows 1 error:

VM95:1 Uncaught SyntaxError: Bad escaped character in JSON at position 82 (line 1 column 83)
    at JSON.parse (<anonymous>)
    at request.onreadystatechange (settings:1:56269)
request.onreadystatechange @ settings:1
XMLHttpRequest.send
getSettings @ settings:1
(anonymous) @ settings:1

bemmbix, 2025-03-10:
My WLAN Password contains a Backslash and I think the parser stumples at this \-char. :-(

IgorYbema, 2025-03-10:
Ah that could be a thing.
Can you paste an example wifi password (not your real one) looking the same as your real wifi password?

IgorYbema, 2025-03-10:
And I'll add this to be fixed in a later version

bemmbix, 2025-03-10:
The password looks like "abc123\xyz"

closed 2025-04-15 by github-actions[bot]