heishamon/HeishaMon #658: Ethernet Port on large Heishamon killed?

Tom54497, 2025-03-12:
After running several month with no issues the large PCB with connected ethernet port isn't online any more. 

With connected ethernet cable the port plugged on Heishamon shows dark solid green and yellow lights on the RJ45 port sides and underneath a red LED. Heishamon now works via WIFI but the landing page of Heishamon shows "Ethernet not installed".

Is that a sign that the ethernet connector is broken?

Tom54497, 2025-03-12:
Yes. Did all that. 

Ordered a new one with Amazon Prime and will report if it works again. Aliexpress would take too much time but if that will happen again I'll get a few spares.

Is there any overview on all the LED colors and their interpretation on the large PCB eg like flashing blue means writing data and so on?

IgorYbema, 2025-03-12:
RED: booting
YELLOW: lost wifi
BLUE blinking on each log line 
Any other suggestions are welcome

Tom54497, 2025-03-13:
Exchanged W5500 and after a reboot it was detected and working. Yellow side LED on port is flashing again. So the port I ordered with Heishamon was broken. It had the manufacturing code 2415.

Tom54497, 2025-11-14:
And once again LAN adapter is broken which ended up in communication loss and error H74. In that case Heishamon setup's wifi isn't accessible any more too. The workaround is demounting ethernet adapter, pressing reset button (maybe several times) and then setup wifi should be accessible via 192.168.4.1 again. Set wifi credentials for your local wifi and blue LED should flickering.

Is there any serial issue with that port HR961160C?

In that case