heishamon/HeishaMon #672: Too many MQTT reconnects

Raszowka, 2025-04-24:
I have been using heishamon along with HA for the past year. No major issues so far.
Same router, same location, just one PCB connected to my heat pump.
The wifi signal is very strong, router is 2 metres far from the heishamon.
Below current status. I have restarted the heatpump at least 5 times during the past days, but the issue is still the same. I get dozens of mqtt reconnections. It's been a couple of weeks now...  Home Assistant gets crazy and it cannot control correctly the heatpump because of this.

A search in the other topics with same issue did not give me any hint

Thanks in advance for your help

[image]

geduxas, 2025-04-24:
Have you tried to restart mqtt server? Also which version do you have? When it start to happening?

Raszowka, 2025-04-24:
Thanks for the reply.
The MQTT servee in HA has been restarted, reinstalled several times, along with backup restore of entire Home Assistant even from 1 month ago, when everything was running smoothly. As I wrote, it started to happen a couple of weeks ago. I do not recall any MQTT update recently. If that's not an issue that has been already raised and you seem to be unaware of any, I will focus on HA and mosquitto then.

Raszowka, 2025-04-29:
Just for anyone else having the same issue. I had to reset to factory settings the heishamon device not once not twice but 5 times before getting it to a working state. Much less MQTT reconnects now (10 in 5 days).

geduxas, 2025-04-30:
It's not makes any sense..

Raszowka, 2025-04-30:
> It's not makes any sense..

as long as the solution worked for me, it does make a lot of sense.

closed 2025-06-05 by github-actions[bot]