heishamon/HeishaMon #674: Heishamon (ESP8266) not stable

maxkr2, 2025-05-11:
Hi Egyras,
I have problems with running Heishamon on own hardware (small image 3.9 on ESP8266) with schematic from here:
[https://oshwlab.com/igor_6537/panasonic-cn-cnt-to-wemos-through-hole](url)

Webinterface is very laggy if connected to my wireless network and also MQTT values are only transmitted around every 30min.
In AP mode everything seems to be fine, website is accesible and reliable.
CRC from heatmpump is ok and values are shown on the home page.

I also tried two different networks (SSIDs): 
main SSID has an 63 character password (there was an issue in the past), this delivered data only once, so I changed to another SSID with a 20 character password. Both passwords does only contain numbers and ASCII-character, no special characters.

If connected to my network, console shows most of the time nothing (blank, black screen), sometimes errors and message 203 bytes received.
Also pages are not accessable at all.

Do you have an idea what's wrong there? 
Is it better to run the large image on an ESP32 because of dual processor and more CPU power? 
Can you give me a hint for wiring serial if that would solve the issue?

Kind regards,
Max

geduxas, 2025-05-11:
Do you compiled from source or flashed already built one?

maxkr2, 2025-05-11:
I flashed the binary. I.e. better to compile it myself?

MiG-41, 2025-05-11:
No , better to use ready binary... Actual 3.9 is quite ok , but for sure you have to ensure best wifi signal ( in stats wifi 100  - it is not the same whatis shown in web..)

maxkr2, 2025-05-11:
Ok, thats fine, I tried to compile on PlatformIO but got some errors (maybe because of missing libs)..

I took my (wired) Access Point in 1m distance to the HeishaMon wifi was fine - but issue persists:
web page is not found, blank console (in web), no settings shown and often loading for minutes on home screen but no data.

geduxas, 2025-05-11:
So it's not firmware related problem's.. next needs to point out where problem lies... Heishamon can run in hot spot mode, try to use that and watch how performs web interface, if it's still laggy, it could lead problems to hardware.. if it's ok, then look in your network, something could be there.

MiG-41, 2025-05-11:
Maybe you are able to connect a serial console to GPIO2 , where debug info should be present. Then it would be possible to know more.

maxkr2, 2025-05-11:
Great hints, thank you!
I tried to debug via GPIO2 and problem seems connecting MQTT-server:
After a connect there is always an output like this:

`received TOP115 Water_Pressure: 1.60
Connecting to mqtt server ...
Heishamon stats: Uptime: 0 days 0 hours...MQTT publish log message failed!
Received bad header. Ignoring this data!
Received bad header. Ignoring this data!
Received bad header. Ignoring this data!
...
Received bad header. Ignoring this data!`

and some entries like this:
`received TOP115 Water_Pressure: 1.60
Received 203 bytes data
Checksum and header received ok!
received XTOP0 Heat_Power_Consumption_Ex
Lost MQTT connection!
Reconnecting to mqtt server ...
Heishamon stats: Uptime: 0 days 0 hours 
Received bad header. Ignoring this data!
Received bad header. Ignoring this data!`

If webpage is working I saw a few MQTT reconnects after hours, but maybe ESP restarts and resets information from time to time?

Also had some WiFi reconnects before, but put AP again in short distance to HeishaMon.
Data collecting from heatpump seems ok.

So questions is, why is it always reconnecting to MQTT?
My MQTT-server needs username an passsword (8 numbers), could this the reason?

krzbor, 2025-05-12:
Check if, after restarting HeishaMon, it isn't starting both STA and AP modes simultaneously.
I had a similar issue once. In the end, I disabled the automatic switch to AP mode at startup

maxkr2, 2025-05-12:
HeishaMon is only running in STA mode (no AP visible), but I also disabled AP mode at startup now. A little bit better, but still MQTT reconnects and connection problems accessing web interface. But for the first time I saw console output in STA mode, but new entries are not appended any more.

Maybe I will try ESP32 with large image. Or as a workaround I could fetch output from debug serial to a raspberry nearby and let it collect information and resend to MQTT server.

maxkr2, 2025-05-15:
Thank you very much for your help.
I tried it again in AP mode, but entering AP was not easy - I had to reflash ESP8266 many times.
Then AP mode was also very instable: Lost connection very often while opening or changing settings.

So I decided to try it an ESP32 and found this "fork" [https://github.com/ElVit/esphome_components/tree/main/components/panasonic_heatpump](url) - because it promises direct HomeAssistant integration and if necessary I can connect it with a W5500 directly to LAN.

Seems, that only I had such problems, so I can close this ticket, of course.

maxkr2, 2025-05-19:
Yes, might be a hardware issue. If I have no luck with my ESP32 I will by the official equipment.

closed 2025-05-19 by maxkr2