heishamon/HeishaMon #675: Heishamon, h74 error (optional PCB,ENABLED)

BenySur, 2025-05-18:
Hello All,

I have been using heidsamon since 2 years without any problems, heishamon is really close to my router with 100% wifi signal and i have no reconnects in mqtt it stays 1 for very long time, i can even notice my homeassistant counter restarting every 49.7 days die to bit issues, i had long operstions like 150+ days without restarts.
My question is when my heisha looses wifi due to something(latest was moving my router) my heatpump goes into h74 error and stops when i close and reestablish the wifi it continues operating. but I'm worries about this happening when i'm not home or when its too cold outside and i domt notice this, so i want to make it bullet proof. 

i dont use too complicated write function i only occationaly use force dhw and monitoring only, my heatpump setup has main board and optional board, for some reason the jnstaller has set it to optional pcb(i believe this is causing the problem according to what i read) enabled and i have some connection on my external compressor and water temp zone 1 sensor on my optional board, i have connected my heisha to cn nmode on my main board.

when i select listen only on my heisha it also gives h74 error, what would your recommendation would be so i do not get h74 error, i can also five up the force dhw option if that creates the problem, or at least if i can get notification of the error and react properly.

appreciate your response

[image]

[image]

[image]

MiG-41, 2025-05-19:
Since you are using real Optional PCB , you should connect heishamon to CN-CNT or CN-CMODE not to HP , but to Optional PCB !

Also you should not use Optional PCB emulation in heishamon as well ( so that is ok ).

BenySur, 2025-05-19:
@MiG-41 thank you for the answer, i will do that now just a question, as i have the optional pcb (bUT i dont have the CZ-TAW1B) do i have to select listen only mode ? or can both be unticked ?

[image]

BenySur, 2025-05-19:
@geduxas @MiG-41 thank you very much, it seems it was just that simple moved it, disconnected my router for 2 minutes nothing happened no error. thank you again, you saved the day

closed 2025-05-19 by BenySur