heishamon/HeishaMon #684: Opentherm with NEST

emilic14, 2025-06-04:
Hello,

Has anyone manage to get Opentherm working with a NEST device ? 
I've seen post of people that have Opentherm working with other devices, but none with a NEST.

And in my configuration I am not being able to communicate with the NEST (regardless of the cable length or connection)
These are my logs regarding OpenTherm: 

`
Wed Jun  4 14:03:36 2025 (412292696): OpenTherm:   Received read slave config
Wed Jun    4 14:03:36 2025 (412292991): OpenTherm: Received DHW set boundaries   request
Wed Jun    4 14:03:36 2025 (412293252): OpenTherm: Request invalid!
Wed Jun    4 14:03:37 2025 (412294214): OpenTherm: Received CH set boundaries   request
Wed Jun    4 14:03:38 2025 (412294508): OpenTherm: Request invalid!
Wed Jun    4 14:03:38 2025 (412295403): OpenTherm: Request invalid!
Wed Jun    4 14:03:39 2025 (412296390): OpenTherm: Read request DHW setpoint
Wed Jun    4 14:03:40 2025 (412296659): OpenTherm: Request invalid!
Wed Jun    4 14:03:41 2025 (412297588): OpenTherm: Read request Max Ta-set   setpoint
Wed Jun    4 14:03:41 2025 (412297849): OpenTherm: Request invalid!
Wed Jun    4 14:03:42 2025 (412298810): OpenTherm: Received status check: 0, CH:   0, DHW: 0, Cooling, 0, OTC: 0, CH2: 0, SWMode: 0, DHWBlock: 0
Wed Jun    4 14:03:42 2025 (412298811): OpenTherm: Send status: CH: 0, Flame:0,   DHW: 0
Wed Jun    4 14:03:42 2025 (412299104): OpenTherm: Request invalid!
Wed Jun    4 14:03:43 2025 (412300032): OpenTherm: Received read slave config
Wed Jun    4 14:03:43 2025 (412300326): OpenTherm: Request invalid!
Wed Jun    4 14:03:44 2025 (412301255): OpenTherm: Received DHW set boundaries   request
Wed Jun    4 14:03:45 2025 (412301549): OpenTherm: Request invalid!
`

I am using the heishamon large version from TheHogNL, running FW 3.9

emilic14, 2025-07-06:
Issue is still there... but no idea how to debug it...

IgorYbema, 2025-08-06:
I will try to get a 2nd hand NEST myself and try to figure out what is wrong.
Can you tell me which NEST model/type you have? I am not familiair with NEST models/types

emilic14, 2025-08-06:
That would be greatly appreciated.
I have a Nest Thermostat 3rd generation.

> 
> Thermostat
> Model : Display 3.4
> Backplate model : Backplate 6.5
> 
> Heatlink:
> Model Amber 2.5

I do have some information and invalid requests, as I shared in my first post, just dont really know what to do with it... 
In the opentherm tab every things seems to be 0 or not connected: 

[image]

emilic14, 2025-11-18:
Yeah, noup... roomTemp is stuck at 20
I'll keep on checking but even if the communication seems a lot better now, even after 2 days the roomTemp has not changed but it is changing on the Nest side.

emilic14, 2026-01-20:
Ok so It seems this is more a Nest issue...
I think I am moving forward with the removal of my Nest and just doing a quick HAS dashboard that will act as the Nest and actually be able to control my heat pump thanks to Heishamon.
But this means no Opentherm...

Anyhow, thanks @IgorYbema for all your help!

closed 2026-02-25 by github-actions[bot]