heishamon/HeishaMon #686: No DS1820 values via MQTT

Stephan2204, 2025-06-04:
DS1820 sensors were read successfully and presentet in the Webportal.
JSON export shows at the end DS1820 and their alias.

But via mqtt no DS1820 topics are visible in the panasonic tree.

MiG-41, 2025-06-05:
Is it ESP8266 or ESP32 ?

For my side they are visible in MQTT:

[image]

Stephan2204, 2025-06-05:
It's the "large PCB v5".   
So it is the ESP32 variant.

There are other values which were not updated. So my vents are turning around with 470rpm since hours and even the frequency (told me mqtt). But Webinterface (and json) looks fine and have actual values. 
Seems that the mqtt values are not updated before sending out.

Short test with a separate USB-C power supply shows no difference in behavior.

Stephan2204, 2025-06-05:
No problem in Heishamon. Problem on chair ;)
Sorry for the issue. 

Regards
Stephan

closed 2025-06-05 by Stephan2204