heishamon/HeishaMon #688: Understand TOPs and SETs

ElVit, 2025-06-06:
Hi, for me it's quite hard to understand which TOP shows which value and which SET controls which setting of my heatpump. 
So I searched in the Panasonic manual. It describes each settings of each menu in different languages.
Then I created an Exel sheet to map each TOP an SET to a menu setting. 

This is the result:
https://docs.google.com/spreadsheets/d/1OvgLmLLbKW66yvIk6bbAhiyKHbbDdxpYG4V99Js4y8w/edit?usp=drivesdk

It's based on a J-series manual.
The mapping is still not complete. 
It would be nice if you guys could help me to complete it by reviewing/commenting it. 
Feel free to link or copy this sheet. 
Maybe it will help others to understand their heatpumps or even decode more TOPs/SETs.

geduxas, 2025-06-07:
Nice work! Just bring it to discussion!

Not all menu functions will be available in top's and set's, because screen is just interface which talks to pump almost in same language as heishamon. So he's you spreadsheet will help a lot to other!

MiG-41, 2025-06-07:
Also there is a point ,how you aproach this.
Since HP is full of TOPS ( some you can change - influence/change by SET , some are measurment ,results , calculation  ). 
In this way also Service cloud works ( and here all names are very similar or identical to Service cloud).

And menu in controler is also interpretation of this TOPs and SETs... 

So better thinking like HP , not like controler's menu ( if you know what i mean...)

MiG-41, 2025-06-07:
For sure Heishamon is not checking the limits , generally you can sent any values , and it is only up to HP if it will accept this value or not.  In this matter also user have to be carefull and have to remember about limits in his automations ( since sometimes HP will simpy not use this value without any feadback , and sometimes some stupid things could apear - i.e. ERROR in too low flow when you set to low water pump speed ).

Also some limits depend's on HP model ,so they are not general.

McMagellan, 2025-06-07:
I've also created a list in German to give you a better overview of the top commands.
Perhaps you can find some more information here for your list. I haven't updated anything in a long time because my rule (for Jeisha) has been finished for five months and works very well.
Thanks to the Heishamon guys, great work.
My list is sorted by top numbers. To find the parameters in the Service Cloud, coordinates for rows and columns are provided for the cloud's parameter table.

[TOP- Liste V2.pdf](https://github.com/user-attachments/files/20640182/TOP-.Liste.V2.pdf)