heishamon/HeishaMon #692: Heat_Power_Production

ReinhardGruber, 2025-06-13:
I’m using Heishamon with a 16 kW T-CAP Aquarea Series M.

Reading live values and controlling the heat pump works flawlessly except for one value:
"TOP15", "Name": "Heat_Power_Production", "Value": 29600, "Description": "Watt"

This value is always between 28000 W and 30000 W, even when the heat pump is turned off.

On the Aquarea remote control, the correct power values are shown in the energy monitor (production, consumption, and COP).

Hardware: Large Board (ESP32)
Connection: LAN (Wi-Fi disabled)

[log.txt](https://github.com/user-attachments/files/20721802/log.txt)

MiG-41, 2025-06-13:
And here is why: https://github.com/Egyras/HeishaMon/discussions/611#discussioncomment-12025868

Time for implementation ... @IgorYbema ?   :)

ReinhardGruber, 2025-06-29:
workaround -> using XTopics

XTOP0 | Heat_Power_Consumption_Extra | 18 | Watt
XTOP1 | Cool_Power_Consumption_Extra | 0 | Watt
XTOP2 | DHW_Power_Consumption_Extra | 0 | Watt
XTOP3 | Heat_Power_Production_Extra | 0 | Watt
XTOP4 | Cool_Power_Production_Extra | 0 | Watt
XTOP5 | DHW_Power_Production_Extra | 0 | Watt

closed 2025-08-04 by github-actions[bot]