heishamon/HeishaMon #693: Still getting 'Received bad header' errors despite checking wiring"

hinge83, 2025-06-27:
Hi everyone,

I've tried every possible wiring combination between my ESP (running Heishamon) and the Panasonic CN-CNT port, but I keep getting this repeated error:
`Received bad header. Ignoring this data!`

Im using af Esp8266 + logic level shifter (3.3V <-> 5V)

I've verified:

- TX and RX are correctly connected (and also tried swapping them just in case)
- GND is properly connected

Still, no valid data is received — just a stream of "bad header" messages.

Any ideas on what else could cause this? Could it be a compatibility issue with my Panasonic model, or is there something else I might have missed?

Thanks in advance for any help!

[image]

MiG-41, 2025-06-27:
Are you using your own compiled firmware ( with modified TX/RX to D6 and D7 ) or ready one from github ?

If ready one , check this topic: https://github.com/Egyras/HeishaMon/issues/683

hinge83, 2025-06-27:
Correct. I unplug wires when booking.

MiG-41, 2025-06-28:
Or do it acc scheme

> Correct. I unplug wires when booking.

Maybe you have wiring connection for this level shifter ? 
Since for "comonly used" you know with side is the Low Voltage ,with High Voltage , an where to put signals , where voltage and GND's ....

Here it is not described .

And at the end you can connect ESP8266 with HP without level shifter. CZ-TAW1 is doing that ( it have inside czip with is 3,3V connecting to HP without any level shifters...).
ESP8266 is able to withstand 5V input ( several confirmations ) , and HP is also able to recognize 3,3v signal from esp8266...

Only problem will be on boot.

hinge83, 2025-06-28:
Thanks. It worked first time I removed the level shifter.
Thank you very much. Looking forward to reestablish integration to HA.