heishamon/HeishaMon #698: No WLAN after AP reboot on HeishaMon communication PCB

longtom99, 2025-07-12:
When there is no WLAN available, HeishaMon communication PCB doesn't reconnect to WiFi.
E.g. i reboot the AP, no re-connection.
I can reproduce it every time.
I have TP-Link APs.

longtom99, 2025-07-17:
Any other rep than i can use instead of this?
Is this rep maintained?

geduxas, 2025-07-19:
you are in right spot, currently heishamon don't have any known wifi issues, are you on latest firmware version? Just in summer time here is quiet enough :)

longtom99, 2025-07-20:
Thanks!
3.9 Firmware, recently bought.

longtom99, 2025-08-09:
@Egyras When are you back to check the issue?

geduxas, 2025-08-09:
I think same was in #621

longtom99, 2025-08-21:
Should be solved...

longtom99, 2025-08-22:
> > Thanks! 3.9 Firmware, recently bought.
> 
> Hey Igor, hey everyone,
> 
> I also bought a HeishaMon large Board on tindie, v5 with Firmware 3.9 installed. I installed a 9W Alexa power Plug with a USB-C cable to ensure faulty reconnects are not related to undervoltage. (Did try two different plugs with different cables.)
> 
> But my HeishaMon large Board looses Connection to the repeater nearly every day. The distance to the repeater is about 2m.
> 
> Today I realized the Board **doesn‘t freeze but goes into AP-mode.**
> 
> Thanks a lot in advance!

Do you have "Enable WiFi hotspot when not connected" enabled?

longtom99, 2025-09-07:
> [@Egyras](https://github.com/Egyras) When are you back to check the issue?

@Egyras Who can have a look at the issue?

longtom99, 2025-09-07:
Every other device is OK, only Heishamon.
It would be easy to send some logfiles to @Egyras or whoever maintains the code.

longtom99, 2025-10-08:
Ok, will try.
But strange that no one else hast this issue.

closed 2026-01-28 by github-actions[bot]