heishamon/HeishaMon #702: Model Aquarea M 9kW

wm1962, 2025-08-04:
Hi teams,
Can you please update the model of Aquarea M, 9kw, 3-ph, T-CAP
ODU: WH-WXG09ME8
IDU:  WH-SDC0316M9E8
KIT : Monoblock

TOP92 | Heat_Pump_Model | 	12 D7 0B 98 14 35 94 0D 83 10 | Model

Thanks and regards,
WM

Handzoff, 2025-08-15:
Hi team,

Can you please update the model of Aquarea 5kW Bi-Block K Serie in the next firmware update?

Indoor unit: WH-SDC0309K3E5
Outdoor unit: WH-UDZ05KE5
KIT-WC05K3E5

Heishamon code (TOP92) Heat_Pump_Model | E2 D5 0C 08 95 02 D6 10 66 95

Thanks in advance!

geduxas, 2025-08-15:
Firmware will no more bring pump model, just PR with update model and that's it.

wm1962, 2025-08-17:
What did you me with "just **PR** with update model and that's it." ?

IgorYbema, 2025-08-17:
Modify https://github.com/Egyras/HeishaMon/blob/master/HeatPumpType.md in your own repo copy and send the update to this repo (a PR is a pull request).

closed 2025-09-22 by github-actions[bot]