heishamon/HeishaMon #705: compiling heishamon

robert-b, 2025-08-16:
Sketch uses 1442959 bytes (110%) of program storage space. Maximum is 1310720 bytes.
Global variables use 59272 bytes (18%) of dynamic memory, leaving 268408 bytes for local variables. Maximum is 327680 bytes.
Sketch too big; see https://support.arduino.cc/hc/en-us/articles/360013825179 for tips on reducing it.
text section exceeds available space in board

Compilation error: text section exceeds available space in board

when using arduino on windows and a esp32s2 dev  board settings.
does some one know which lib consumes the most flash space?
i am just downloading the latest versions.

IgorYbema, 2025-08-16:
The code isn't tested for the esp32s2. Pinouts may be different.
But try to choose another filesystem layout to have more space for the sketch.

robert-b, 2025-08-16:
[image]

closed 2025-08-16 by robert-b

IgorYbema, 2025-08-17:
Yes you could make your own partition like that.
Heishamon pre-builds are using the min_spiffs option: https://github.com/espressif/arduino-esp32/blob/master/tools/partitions/min_spiffs.csv