heishamon/HeishaMon #718: Add New Panasonic WH-SDC0309K3E5/WH-UDZ07KE5 Model to HeatPumpType.md

kamilzierke, 2025-09-29:
# GitHub Issue: Add New Panasonic WH-SDC0309K3E5/WH-UDZ07KE5 Model to HeatPumpType.md

## Title
**Add new Panasonic WH-SDC0309K3E5/WH-UDZ07KE5 model (K-series, 7kW) with updated byte signature**

## Description
I would like to add a new variant of the Panasonic heat pump model to the `HeatPumpType.md` database. This model is functionally identical to existing model #56 but has a slightly different byte signature, representing a newer version of the same hardware.

## Model Information
- **IDU:** WH-SDC0309K3E5
- **ODU:** WH-UDZ07KE5
- **KIT:** KIT-SDC07KE5
- **Rated Power:** 7 kW
- **Series:** K-series
- **Refrigerant:** R32
- **Type:** HP (High Performance)

## Byte Signature (bytes 129-138)
The heat pump identification bytes from the protocol are:
```
E2 D5 0D 08 95 02 D6 10 67 95
```

## Comparison with Existing Model
The closest existing model (ID #56) has the signature:
```
E2 D5 0B 08 95 02 D6 0F 67 95
```

**Differences:**
- Byte 131: `0D` vs `0B` (newer version)
- Byte 136: `10` vs `0F` (newer version)

## Electrical Configuration
- **Heat Pump Power Supply:** 1-phase
- **Internal Heater Power Supply:** 1-phase (separate phase)
- **Internal Heater Capacity:** 3 kW
- **Total Electrical Configuration:** 2-phase setup

This configuration is common in Polish installations where the heat pump runs on one phase and the backup heater on another.

## Proposed Addition to HeatPumpType.md
Please add the following entry to the `HeatPumpType.md` file:

```markdown
| 57 | E2 D5 0D 08 95 02 D6 10 67 95 | WH-SDC0309K3E5 | WH-UDZ07KE5 | KIT-SDC07KE5 | 7 | 1ph* | HP | *2-phase configuration with 3kW heater on separate phase |
```

## Additional Information
- **User Verification:** Confirmed by purchase invoice
- **Operational Status:** Successfully decoding all protocol parameters
- **System Status from Logs:**
  - Mode: Heating + DHW (byte 6 = 62)
  - Outdoor Temperature: 6°C
  - DHW Current: 41°C / Target: 45°C
  - Compressor Frequency: 47 Hz
  - No errors detected

## Testing Data
Developers can use this log fragment for testing (full response available if needed):
```
71,C8,01,10,56,55,62,49,00,55,00,25,...,E2,D5,0D,08,95,02,D6,10,67,95,...
```

This model is fully compatible with the existing protocol decoding and only requires addition to the model identification database.

## Labels
- `new-model`
- `protocol`
- `k-series`

Thank you for considering this addition to HeishaMon!

## Full log
```
[20:31:21.508][I][panasonic_heatpump:034]: >>> extra_polling_request[111]
[20:31:21.514][I][panasonic_heatpump:047]: >>> 71,6C,01,21,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,
[20:31:21.527][I][panasonic_heatpump:047]: >>> 00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,
[20:31:21.545][I][panasonic_heatpump:047]: >>> 00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,01

[20:31:21.969][I][panasonic_heatpump:034]: <<< extra_polling_response[203]
[20:31:21.990][I][panasonic_heatpump:047]: <<< 71,C8,01,21,8A,EA,01,00,00,00,00,00,00,00,67,04,01,00,01,00,67,18,01,00,01,00,01,00,00,00,01,00,00,00,01,00,00,00,01,00,
[20:31:22.007][I][panasonic_heatpump:047]: <<< 00,00,01,00,01,00,01,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,
[20:31:22.007][I][panasonic_heatpump:047]: <<< 00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,
[20:31:22.017][I][panasonic_heatpump:047]: <<< 00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,
[20:31:22.029][I][panasonic_heatpump:047]: <<< 00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,
[20:31:22.041][I][panasonic_heatpump:047]: <<< 00,00,3B

[20:31:26.587][I][panasonic_heatpump:034]: >>> polling_request[111]
[20:31:26.587][I][panasonic_heatpump:047]: >>> 71,6C,01,10,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,
[20:31:26.587][I][panasonic_heatpump:047]: >>> 00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,
[20:31:26.587][I][panasonic_heatpump:047]: >>> 00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,12

[20:31:26.995][I][panasonic_heatpump:034]: <<< polling_response[203]
[20:31:26.996][I][panasonic_heatpump:047]: <<< 71,C8,01,10,56,55,62,49,00,55,00,25,00,00,00,00,00,00,00,00,59,15,11,55,56,1D,55,55,55,19,56,00,00,00,00,00,00,00,83,80,
[20:31:27.012][I][panasonic_heatpump:047]: <<< 80,80,AD,71,71,97,99,00,00,00,00,00,00,00,00,00,00,00,80,85,5B,8A,85,85,D0,7B,78,1F,7E,1F,1F,79,79,8D,8D,9E,94,7B,8F,B7,
[20:31:27.012][I][panasonic_heatpump:047]: <<< A3,7B,8F,94,85,80,8F,8A,94,9E,8F,8A,94,9E,85,8F,8A,04,5B,78,C1,0B,7E,7C,1F,7C,7E,00,00,00,55,55,55,01,11,15,59,05,1C,12,
[20:31:27.021][I][panasonic_heatpump:047]: <<< 65,21,51,15,01,54,80,80,80,E2,D5,0D,08,95,02,D6,10,67,95,9D,00,A9,86,98,9D,32,32,9C,AD,32,32,32,83,9C,9D,BF,96,99,80,00,
[20:31:27.033][I][panasonic_heatpump:047]: <<< 80,00,00,5D,01,19,30,00,00,2F,14,30,5F,37,01,39,01,01,01,AF,00,00,69,01,00,01,00,00,01,00,00,08,01,00,00,00,00,00,00,03,
[20:31:27.046][I][panasonic_heatpump:047]: <<< 80,33,41```

geduxas, 2025-10-01:
ok, we are going somewhere :D 

| | | IDU  |  | ODU  |
|--|------------|------------|------------|------------|
1|D5E1| 95080A| D601| 95670E
2|D5E1| 95080B | D601| 95670C
3|D5E1| 95080B|D601| 95670F

@kamilzierke have different Indoor unit code (IDU) comparing to @longtom99 and @hoschiking , so maybe there is some revision's on PCB? sticker's? or something? maybe you could share your indoor unit PCB photo's ? maybe assemble locations different? or maybe build dates? ODU (outdoor units are all different..)

geduxas, 2025-10-31:
Thanks, data matrix (code box) contains serial number, so you should hide it too if you hiding serials :) truly don't know are serials have any reason to hide, they are in public space mostly :D 

Also i don't remember do, internal unit have any manufacturing date on it? I don't see anything on label.. 

What i am looking it could be related to manufacture date.. or just main board revision numbers.. so outdoor unit labels ar not what i am looking for :)